DTUSDC/ros #9: Build ROS driver for IMU

krush3r, 2020-01-26:
The IMU used is Sparton [AHRS-8P](https://www.spartonnavex.com/product/ahrs-8p/). 

## Todo:
- build a encoder to read data from IMU
- publish relevant output to a ROS topic

obviyus, 2020-01-28:
Study the following standards before implementing:
- [REP 103](https://www.ros.org/reps/rep-0103.html)
- [REP 105](https://www.ros.org/reps/rep-0105.html)

krush3r, 2020-02-05:
Driver for IMU (Sparton ahrs-8p) ready and now publishes:-
- Orientation
- angular velocity
- linear acceleration

Using code base from: https://github.com/RoboticsClubatUCF/ucf_AHRS

krush3r, 2020-02-12:
Improved output data format which now publishes
- Orientation in degrees
- Angular velocity in radian/sec
- linear acceleration in m/s^2

krush3r, 2020-02-24:
IMU driver made according to the standards of
- [REP 103](https://www.ros.org/reps/rep-0103.html)
- [REP 105](https://www.ros.org/reps/rep-0105.html)

closed 2020-02-24 by krush3r