DTUSDC/ros #12: Fine tuning Navigation Stack

obviyus, 2020-02-05:
Starting reference for tuning navigation stack:

- http://kaiyuzheng.me/documents/navguide.pdf
- [ROS Wiki Navigation Tuning](http://wiki.ros.org/navigation/Tutorials/Navigation%20Tuning%20Guide)

obviyus, 2020-02-19:
Somewhat desirable simulations obtained after pushing 3bb44642. Still need to work on tweakign recovery behaviors and goal tolerances.

obviyus, 2020-02-23:
Lots of improvements in commits 8a79763 and 80ebdc7:
- Updating gmapping intervals has fixed frequency of map updates, vehicle no longer hits unknown obstacles
- Increasing inflation radius avoids "wall hugging"

tanmayj2020, 2020-03-11:
Fine still done but it us not perfect

obviyus, 2020-03-12:
Investigate more SLAM techniques. Here's a paper evaluating a few 2D slam techniques: https://core.ac.uk/download/pdf/29175747.pdf